Zinc-OS/continuousprint #13: Feature request: Counter for the whole queue or for single file

TylonHH, 2021-01-01:
Hey,
I really love the counter. Is it possible to add an counter for the whole queue?
For example:
**Option 1**
_Queue
File A count: 3
File B count: 5
File C count 2_

**Option 2**
_Queue count: 6
File A
File B
File C_

In the second option I'll get 6x A. B and C. That would help alot.

Cheers

TylonHH, 2021-03-03:
Reopen 😎

toddmedema, 2022-01-12:
Hey @TylonHH - seems like this is now implemented? I'm seeing counts in my plugin 
[image]

TylonHH, 2022-01-13:
No this is not new. I suggested an counter for the whole queue, not for every single file.

Zinc-OS, 2022-01-13:
It's not a counter for the whole queue, just per file, same as before.

TylonHH, 2022-01-13:
> It's not a counter for the whole queue, just per file, same as before.
Yeah I know, but I request this feature to set an counter for the loop instead for each file.

toddmedema, 2022-01-13:
@Zinc-OS thoughts on adding a total count as part of total time? https://github.com/Zinc-OS/continuousprint/issues/59

Maybe something like:
`Time remaining: 30 minutes on current part, 7 hours on queue (1/7 parts printed)`

closed 2022-03-17 by Zinc-OS