Zinc-OS/continuousprint #21: Wont Work on Oktopi Version 1.5.3

Floslis, 2021-02-08:
Teh plugin stop after the first print

Zinc-OS, 2021-02-10:
With the default bed-clearing script, it does wait a long time as it cools to 25 degrees Celsius. If the room temperature is higher than 25 degrees, than it will never finish.

jaykav99, 2021-02-16:
I am currently having the 25 degree issue...any chance a future update will allow a editable value before starting the print, so we can push it up just a few degrees?

Zinc-OS, 2021-02-19:
In the settings, you can already change the value:
```
M17 ;enable steppers
nM91 ; Set relative for lift
nG0 Z10 ; lift z by 10
nG90 ;back to absolute positioning
nM190 R25 ; set bed to 25 for cooldown<--Set the R25 to the temperature you want. At 45 degrees that would be R45
nG4 S90 ; wait for temp stabalisation
nM190 R30 ;verify temp below threshold<--Set this value to your temperature to about the temperature you want it plus five, or you can just delete this line. At 45 degrees this would be R50
nG0 X200 Y235 ;move to back corner
nG0 X110 Y235 ;move to mid bed aft
nG0 Z1v ;come down to 1MM from bed
nG0 Y0 ;wipe forward
nG0 Y235 ;wipe aft
nG28 ; home
```
Also, don't forget to press save, or it will not be set.

jaykav99, 2021-02-19:
Perfect sorry was some how not seeing where to edit it...just me being blind.

Zinc-OS, 2021-02-19:
No problem

jones3511, 2021-02-26:
do you need to have the "n" in the script.  I was running a script for a couple weeks that worked just fine and after the update the script, that hasn't changed is running into the print and causing the printer to error out.

Zinc-OS, 2021-02-26:
> do you need to have the "n" in the script. I was running a script for a couple weeks that worked just fine and after the update the script, that hasn't changed is running into the print and causing the printer to error out.

Sorry about that. it was part of /n. That would indeed cause an error.

closed 2021-03-13 by github-actions[bot]