Zinc-OS/continuousprint #22: Problems with clearing script after update

jones3511, 2021-02-25:
I dont know why im having issues all of a sudden by both of my printers that are using a bed clearing script in the past have both started running into the print and cause the printer to shut down due to binding issues.  I am also having issues with the printer waiting until the bed temperature reaches it's cooling temp.  It always tries to clear 5 to 10 degrees hotter than what is assigned.

M140 S0 ; set bed temp

M17 ;enable steppers

G91 ; Set relative for lift

G0 Z30 ; lift z by 30

G90 ;back to absolute positioning

M106 ; turn fans on

G0 X0 Y235

G0 X75

G0 Y230

G0 Z10v ;come down to .1MM from bed

G4 S600 
M300 S400 P1000
G4 P1200 ; Pause for 900ms
M300 S400 P1000

M140 S35 ; set bed temp
M190 R35 ; wait for bed temp

M140 S35 ; set bed temp
M190 R35 ; wait for bed temp

M140 S35 ; set bed temp
M190 35 ; wait for bed temp

M190 R35 ;verify temp below threshold

M300 S400 P900
G4 P1200 ; Pause for 900ms
M300 S400 P900
G4 P1200 ; Pause for 900ms
M300 S400 P900
G4 P1200 ; Pause for 900ms
M300 S400 P900
G4 P1200 ; Pause for 900ms
M300 S400 P900

G0 X0 Y230 ;move to back corner

G0 Z10v ;come down to .1MM from bed

G0 Y0 F6000 ;wipe forward

G0 X185 Y0

G0 x0 Y230

G0 X215

G0 Y0

G0 x0 Y0

G0 X200 

G0 X0 Y230

G0 Z25 ; lift z by 10

M300 S400 P400
G4 P800 ; Pause for 400ms
M300 S400 P400

Zinc-OS, 2021-02-26:
The reason the printer decides its finished 5-10 degrees higher is that its main job is to _heat up_, not cool down.
It only needs to make sure the plastic is hot enough to melt, and not to hot to burn. If the temperature is at 240 C where it should actually be at 230, waiting for it to drop 10 degrees is a waste of precious time. In fact, the printer will often wait too long, and if it drops below the wanted temperature, it will have to wait until it reaches that temperature again.
The easiest solution is to set the temperature 5-10 degrees lower than what you want it to be. 

Other printers(like mine), will wait until the temperature is exactly what you set it too. That can lead to the printer taking way too long to cool(It will wait until it is 25 C even if it is at 25.1 C). In the latter you want to set it ~5 degrees hotter, in the former you want 5-10 degrees cooler.

There is no universal bed clearing command, and the difference between printers goes beyond hardware(like size) to differences in the firmware(In this case how temperature commands are handled).

jones3511, 2021-02-27:
> The reason the printer decides its finished 5-10 degrees higher is that its main job is to _heat up_, not cool down.
> It only needs to make sure the plastic is hot enough to melt, and not to hot to [...]

Alright I get the temp clearing fix and will give it a try, but i am still having some major issues with the clearing commands.  For some reason my printer is trying to home after a print.  It is really hard to pin down why it is doing this because it seems to be random.  For instance I had 7 prints in the que to print overnight.  I came back in the morning and it had printed 4 with no issue.  After the 5th print finished the printer tried to home on top of my print, this ended up shutting the printer down.  But I have had it print 1 and clear it just fine and then tries to home on the second.

M17 ;enable steppers
M91 ; Set relative for lift
G0 Z10 ; lift z by 10
G90 ;back to absolute positioning
M106 ; Turn on fans
G0 X0 Y235 ;move bed all the way forward
G0 X75 ; move printhead close to print for cooling
G0 Y230 ; move printhead close to print for cooling
G0 Z10v ; come down to 1mm from bed
M300 S400 P3000 ; long beep for dwell command
G4 S1200 ; dwell for 20 mins
M300 S400 P3000 ; long beep dwell is over
G4 P1200 ; beep pause
M300 S400 P1500 ; short beep dwell is over

M140 S35 ; set bed temp
M190 R35 ; wait for bed temp
M140 S35 ; set bed temp
M190 R35 ; wait for bed temp
M140 S35 ; set bed temp
M190 R35 ; wait for bed temp

M300 S400 P900
G4 P1200 ; Pause for 900ms
M300 S400 P900
G4 P1200 ; Pause for 900ms
M300 S400 P900
G4 P1200 ; Pause for 900ms
M300 S400 P900
G4 P1200 ; Pause for 900ms
M300 S400 P900
G0 X0 Y230 ;move to back corner
G0 Z10v ;come down to 1MM from bed
G0 Y0 F6000 ;wipe forward
G0 X185 Y0 ; wipe across
G0 x0 Y230 ; move bed all the way forward
G0 X215 ; printhead halfway across
G0 Y0 ; wipe forward again
G0 x0 Y0 ; home x y 
G0 X200 ; wipe across again
G0 X0 Y230 ; wipe forward again
G0 Z10 ; lift z by 10

M300 S400 P800
G4 P800 ; Pause
M300 S400 P1200

G28 ; home

Zinc-OS, 2021-02-27:
Alright, I'm going to label this as a bug.

You did say this only happened after you updated?

jones3511, 2021-02-27:
> Alright, I'm going to label this as a bug.
> 
> You did say this only happened after you updated?

correct

jones3511, 2021-02-27:
Send: N43450 G1 X93.942 Y113.098 E2403.04398*96
Recv: ok
Send: N43451 G1 X84.015 Y123.025 E2407.2464*95
[...]
Recv: ok
Send: N43452 G1 X84.015 Y121.328 E2407.75439*104
Recv: ok
Send: N43453 G1 X93.942 Y111.401 E2411.95681*99
Recv: ok
Send: N43454 G1 X93.942 Y109.704 E2412.46479*99
Recv: ok
Send: N43455 G1 X84.015 Y119.631 E2416.66721*97
Recv: ok
Send: N43456 G1 X84.015 Y117.934 E2417.17519*104
Recv: ok
Send: N43457 G1 X93.942 Y108.007 E2421.37761*105
Recv: ok
Send: N43458 G1 X93.942 Y106.309 E2421.88589*101
Recv: ok
Send: N43459 G1 X84.015 Y116.237 E2426.08853*97
[...]
Recv: ok
Send: N43460 G1 X84.015 Y114.539 E2426.59681*101
Recv: ok
Send: N43461 G1 X93.392 Y105.163 E2430.56618*109
Recv: ok
Send: N43462 G1 X91.695 Y105.163 E2431.07417*102
Recv: ok
Send: N43463 G1 X84.015 Y112.842 E2434.32515*97
Recv: ok
Send: N43464 G1 X84.014 Y111.566 E2434.70711*111
Recv: ok
Send: N43465 G1 X84.014 Y111.146 E2434.83283*106
[...]
Recv: ok
Send: N43466 G1 X89.998 Y105.163 E2437.36584*102
Recv: ok
Send: N43467 G1 X88.301 Y105.163 E2437.87382*102
Recv: ok
Send: N43468 G1 X84.013 Y109.45 E2439.68886*92
Recv: ok
Send: N43469 G1 X84.011 Y107.756 E2440.19594*98
Recv: ok
Send: N43470 G1 X86.604 Y105.163 E2441.29365*98
Recv: ok
Send: N43471 G1 X84.907 Y105.163 E2441.80163*106
Recv: ok
Send: N43472 G1 X84.009 Y106.06 E2442.18157*90
Recv: ok
Send: N43473 G1 X84.009 Y105.567 E2442.32914*109
Recv: ok
Send: N43474 G1 X84.009 Y105.163 E2442.45008*110
Recv: ok
Send: N43475 G1 F3000 E2441.45008*61
[...]
Recv: ok
Send: N43476 M140 S0*103
Recv: ok
Send: N43477 M107*38
Recv: ok
Send: N43478 G91*29
Recv: ok
Send: N43479 G1 E-2 F2700*60
Recv: ok
Send: N43480 G1 E-2 Z0.2 F2400*111
Recv: ok
Send: N43481 G1 X5 Y5 F3000*70
Recv: ok
Send: N43482 G1 Z10*90
Recv: ok
Send: N43483 G90*24
Recv: ok
Send: N43484 G1 X0 Y235*34
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: ok
Send: N43485 M106 S0*105
Recv: ok
Send: N43486 M104 S0*104
Recv: ok
Send: N43487 M140 S0*105
Recv: ok
Send: N43488 M84 X Y E*120
[...]
Recv: ok
Send: N43489 M300 S1000 P400*20
Recv: ok
Send: N43490 M300 S0 P1000*24
Recv: ok
Send: N43491 M300 S1500 P400*24
Recv: ok
Send: N43492 M300 S0 P1000*26
Recv: ok
Send: N43493 M300 S2000 P400*28
Recv: ok
Send: N43494 M82*23
Recv: ok
Send: N43495 M104 S0*106
Recv: ok
Changing monitoring state from "Printing" to "Finishing"
Send: N43496 M400*43
Recv: ok
Changing monitoring state from "Finishing" to "Operational"
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: N1 M82*24
Recv: ok
Send: N2 M140 S65*84
Recv: ok
Send: N3 M104 S294 &#65279*115
Recv: ok
Send: N4 M300 S1000 P400*18
Recv: ok
Send: N5 M280 P0 S160*40
Recv: ok
Send: N6 G4 P100*106
Recv: //action:notification Sleep...
Recv: ok
Send: N7 G28*20
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Printing" to "Error: Printer halted. kill() called!"
Send: M112
Send: N8 M112*41
Send: N9 M104 T0 S0*40
Send: N10 M140 S0*84
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
Connection closed, closing down monitor

Zinc-OS, 2021-02-27:
Since this presumably has to do with the update to 1.2.1, you can use this url to install the previous version for now:
`https://github.com/Zinc-OS/continuousprint/archive/1.2.0b.zip`

jones3511, 2021-02-28:
> Since this presumably has to do with the update to 1.2.1, you can use this url to install the previous version for now:
> `https://github.com/Zinc-OS/continuousprint/archive/1.2.0b.zip`

I reverted back to the previous version and it is working so far.

Zinc-OS, 2021-02-28:
I've changed where the bed clearing script is executed to where it was in 1.2.0, so this should fix it:
`https://github.com/Zinc-OS/continuousprint/archive/dev.zip`

AndiP1234, 2021-04-29:
G0 Z10**v** ;come down to 1MM from bed
I think that the v at the end of the G0 code is an issue. I found the v also in the default bed clearing script.

https://marlinfw.org/docs/gcode/G000-G001.html

jones3511, 2021-05-04:
I just updated from the 1.2.0 build.  Still does not run clearing script before homing and starting the next print.  I dont know what the issue, but I'm just going to continue to use the 1.2.0 build