Zinc-OS/continuousprint #27: Bed homing after a print instead of clearing the bed

ryzerman, 2021-04-02:
Hi

I am having a problem clearing the bed after a print.

For some reason the printer is trying to home after the print and coaching in to the print, instead of running the clearing script.

I had this working fine a few months ago but cant seem to get it to work now and I have 108 identical parts to print.

The printer is an ender-3 running a SKR Mini E3 V2 and a BL Touch (I dont know if that makes a difference)

The script is below:

M17 ;enable steppers
G91 ; Set relative for lift
G0 Z10 ; lift z by 10
G90 ;back to absolute positioning
M190 R25 ; set bed to 25 for cooldown
G4 S90 ; wait for temp stabalisation
M190 R30 ;verify temp below threshold
G0 X200 Y235 ;move to back corner
G0 X110 Y235 ;move to mid bed aft
G0 Z1v ;come down to 1MM from bed
G0 Y0 ;wipe forward
G0 Y235 ;wipe aft
G28 ; home

The Cura finishing script is:
G91; set coordinates to relative
G1 F1800 E-3; retract
G1 F3000 Z10; lift nozzle off the print 10mm
G90; change to absolute
G1 X0 Y220 F1000 ; prepare for part removal
M106 S0 ; turn off cooling fan
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors

I really hope you can help as I really need to automate printing these as if I am not running the printer overnight I won't get it printed in time.

Kind Regards
Ryan

Zinc-OS, 2021-04-02:
Could you upload logs?

Zinc-OS, 2021-04-02:
Also Octoprint and plugin version, and if you've updated this plugin or Octoprint recently.

ryzerman, 2021-04-05:
Octoprint
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0.3 Safari/605.1.15
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 4041732096
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin bugfix-2.0.x (Jan 11 2021

ryzerman, 2021-04-05:
Continuous Print (1.2.2)

ryzerman, 2021-04-05:
Where do I find logs?