Zinc-OS/continuousprint #34: API Call to add File to queue

Peinbauw, 2021-06-02:
Hi to all, 

i like this Plugin very much, i am using it in my printer farm for all printers.
Is there a way to add files to the queue by an API Call or can it be added to the plugin?

Greetings Wolfgang
3D Maker Lab

hughht5, 2021-06-11:
+1 for this - I'm trying to add files direct from my store. Is there a plan for this? or a sensible way of adding it?

Peinbauw, 2021-07-02:
Hi hughht5, 

in the meantime i found out, this can be done directly by using the default functions of octoprint in your OS Command Line.

Example:
`curl -k -H "X-Api-Key: <your key>" -F "name=test.gcode" -F "path=test.gcode" -F "sd=false" -F "count=1" "http://192.168.0.107/plugin/continuousprint/addqueue"`

The File "test.gcode" has to be uploaded before, which can also be done by a REST API call

`curl -k -H "X-Api-Key: <your key>" -F "select=false" -F "print=false" -F "file=@C:\temp\test.gcode" "http://192.168.0.110/api/files/local"`

Greetings Wolfgang

closed 2021-07-02 by Peinbauw