Zinc-OS/continuousprint #48: Is this plugin abandoned?

smartin015, 2021-11-10:
Hi Louis,

I sent an email ~10 days ago asking for your thoughts on upstreaming a substantial refactor I made in [my fork of your plugin](https://github.com/smartin015/continuousprint), after opening up #41 in mid-July as a heads up that I'd be doing some work to incorporate further print automation. I haven't heard back from you in either case. 

Every other open issue indicates you're not actively working on this plugin:

* #44, opened 2021-09-11 and never responded
* #43, opened 2021-07-20 and never responded
* #42, opened 2021-07-15 and never responded
* #33, opened 2021-05-25 and autoclosed twice with no response
* #27, last interaction 2021-06-12 (reopen from autoclose)
* #22, last interaction 2021-02-28
* #13, last interaction2 2021-03-02 (reopen from autoclose)

I'd be happy to adopt the plugin and continue to make improvements, including the ones I've already made in my fork:

* Status of continuous print automation describing current state
* Queue items expand to show individual item completion state, number of retries, time it took etc.
* Triple-dot menu allows clearing all prints or just the successful ones (allowing for retries of failed prints)
* File list is moved up and collapsable, so that clicking to add a file to the queue doesn't cause the whole UI to shift down
* Split out logic for the print queue into a separate python module, and unit tested it
* Split out business logic of handling print actions into a separate module and unit tested it
* The frontend relies more upon bootstrap.js for dynamic value updates, so JS doesn't need to be injected into the DOM as a string (simpler frontend code.
* Better handling of the spaghetti detective plugin for automated failed print detection - if SD pauses a print before a configurable time limit, the print restarts (else it pauses as usual and waits for manual input)
* There is a limit on retries, so if a print fails repeatedly (maybe because it's not set up correctly for good bed adhesion) it'll move on to the next print after clearing the bed again.

Please let me know your thoughts - **if I don't hear back in another 10 days (i.e. by 2021-11-20) I'll open an [abandoned plugin ticket](https://github.com/OctoPrint/plugins.octoprint.org/issues/new?template=abandoned_plugin.yml) and begin the process of migrating ownership**. If you'd prefer to keep ownership but would prefer not to merge in my changes, I'll gladly publish my edits as a new plugin.

smartin015, 2021-11-22:
I've opened the ticket at https://github.com/OctoPrint/plugins.octoprint.org/issues/966.

smartin015, 2021-12-12:
Just a friendly update - the plugin adoption process seems to be pretty slow. I've been using it for my own purposes for awhile, and it's quite stable.

Not too many changes of late, except for adding yellow highlights when queue entries change (to track the impact of queue ordering actions, deletions etc) and some [unit tests to ensure compatibility](https://github.com/smartin015/continuousprint/blob/master/continuousprint/print_queue_test.py#L89) with the old queue data format.

Zinc-OS, 2021-12-13:
Not quite.

smartin015, 2021-12-13:
> Not quite.

Could you please clarify? Would you prefer to

- Keep plugin ownership and pull in my changes?
- Keep plugin ownership and I'll split off as a separate plugin?
- Hand off ownership via https://github.com/OctoPrint/plugins.octoprint.org/issues/966 ?

Zinc-OS, 2021-12-13:
If I were to keep ownership, I would pull in your changes(so a), as you fixed the UI, as well as the redundancies caused by the print-count(new queue items are added to the count of an item if it already exists, which makes sense.)

smartin015, 2021-12-21:
Sounds like a plan. I've opened #51 and will obsolete https://github.com/OctoPrint/plugins.octoprint.org/issues/966 once it's merged.

PR #51 (2021-12-24): Continuous print overhaul
Closes #48

Apologies in advance for the large dump of code - please let me know if you need anything to more easily review it.

A couple notes to guide review:

* `print_queue.py` is a pretty basic queue implementation that stays in sync with JSON storage
* `driver.py` manages the print queue by preparing printer actions whenever certain `on_*` event methods are called
* `__init__.py` now mostly just forwards `on_event` events to `driver.py`,  although it does also provide the web API access to the print queue
* Tests are runnable directly (e.g. `python3 driver_test. [... cut]
closed 2021-12-24 by Zinc-OS