Zinc-OS/continuousprint #53: After Update it doesnt work anymore at all

davidgermany, 2021-12-26:
[image]

[image]
[octoprint (1).log](https://github.com/Zinc-OS/continuousprint/files/7776515/octoprint.1.log)

smartin015, 2021-12-29:
Hey @davidgermany - apologies for the trouble. The latest version seems to be having some trouble parsing the SD field from old prints. 

I've opened #55 which I believe should correctly parse the older saved data from prior versions of the plugin. You can try to pull/use that, or if you'd rather just get on with things you can delete the plugin data (see the little eraser icon) and it should behave after a restart.

[image]

In any case, please report back if there's still errors / logs exceptions and I'll make another pass.

smartin015, 2022-01-22:
@davidgermany, are you stlil having issues or can this be closed out?

smartin015, 2022-03-13:
If you have any further problems, you're welcome to open a new issue at https://github.com/smartin015/continuousprint/issues. Thanks!

davidgermany, 2022-03-13:
Yeah, I just re-installed octopi and the newer version of your plugin. It
works again

Thank you, you can close this "ticket"

closed 2022-03-17 by Zinc-OS