Zinc-OS/continuousprint #54: plugin does not work after update

sidneyns, 2021-12-28:
Hello 

The plugin is not working anymore, it is loading the continuous print script when you start the managing procedure to start and the printer goes crazy. 

THis is the script I have bellow. What ishappening is that the printer goes to 28 degress, kinda kicks the part ( that is not there), but in the process it mess up the printer motor and belts trying to run on a coordanate that does not exist 

M17 ;enable steppers
G91 ; Set relative for lift
G0 Z10 ; lift z by 10
G90 ;back to absolute positioning
G0 X30 Y1 F3600
M190 R28 ; set bed to 25 for cooldown
M190 R28 ; set bed to 25 for cooldown
M190 R28 ; set bed to 25 for cooldown
M190 R28 ; set bed to 25 for cooldown
M190 R28 ; set bed to 25 for cooldown
M190 R28 ; set bed to 25 for cooldown
G0 X30 Y255
G0 Z1v
G0 Y0
G0 X30 Y255
G0 X100 Y255
G0 Y0
G0 X100 Y255
G0 X160 Y255
G0 Y0
G0 X160 Y255
G0 X220 Y255
G0 Y0
G0 Y235
G28 ; home

TylonHH, 2021-12-28:
https://github.com/Zinc-OS/continuousprint/issues/53

Zinc-OS, 2021-12-29:
Different issue.
That has to do with the queue.
This happens because the bed clearing script is run before the completion of any prints, and the nozzle could be basically anywhere. Because his script starts in relative positioning, it could attempt to move the nozzle out of bounds.
I've made sure that the script is only run after prints have been completed, which should fix this.

sidneyns, 2021-12-29:
I saw that there is now a new script area on the plugin, should I just
change the script I created to the other part?

Zinc-OS, 2022-01-01:
Sorry for the delay.
By the other part you mean the section under the bed clearing script?
that's for after the entire queue has been completed, in case you need to run something after all print jobs have been finished, perhaps nozzle cleaning. If you remove your script to that part you would have more problems, as it wouldn't run at all.

If you update the plugin, this should be fixed because the script will now only be run after the first print completion(not before).

sidneyns, 2022-01-01:
thank you great job by the way.

one thing I noticed is that now if I decide to cancel the job, I need to
cancel both ways ( on continuous and also on the print job) , just getting
used to it

jones3511, 2022-01-03:
Im not sure what is going on with it as of time of update.  It never stops initializing. I cant add files to the que and the start managing button does do anything? Was working without issue before update.

TylonHH, 2022-01-03:
> It never stops initializing.
Are tried this with the newest version [1.3.1](https://github.com/Zinc-OS/continuousprint/releases/tag/1.3.1)?

sidneyns, 2022-01-03:
unistall old version and reinstall that it works

closed 2022-01-05 by Zinc-OS