Zinc-OS/continuousprint #56: loop missing in the newest version 1.3.1

TylonHH, 2022-01-03:
As the newest version works, I'm missing the loop button.

jones3511, 2022-01-14:
As am I.  I will be using older version until its implemented.  Great work devs on making improvements but removing something as big as that is a big oversight IMO.

smartin015, 2022-01-22:
Now that individual queue item states are tracked, having a "loop" button is a bit less obvious of an implementation. Do y'all use the loop button in a way that couldn't be done by just setting the number of copies to a high number (e.g. 100)? If so, could you give some more details?

I had a couple ideas on how to implement it, but I haven't decided as I usually don't want to continue printing things indefinitely.

TylonHH, 2022-01-22:
Sure. I just use the loop function. For example, I am printing a box with one body and one lid. So I don't want to print 100 bodies and then 100 lids. I would rather print bodies and lids one after the other. I also don't care how many bodies and lids are printed. I print until the filament is empty. But it would be cool if I could specify the number of loops.

smartin015, 2022-02-07:
Hey @TylonHH, @jones3511 - I think I landed on a good solution for looping. It's currently available on the master branch of my fork (see https://github.com/smartin015/continuousprint/issues/3) while I wait for @Zinc-OS to review #62 and get it properly merged into the main plugin repo. 

[image]

Essentially, you can now group individual queue items (i.e. a print file with a number of repeats) into "jobs" (a list of queue items with a number of repeats). Under this scheme, you can add a box gcode file and a lid gcode file to the same job, then set the number of runs of the job, and it should interleave prints of each of these parts so you'll get box, then lid, then box etc.

The cool part about this is that now you can have multiple Job "loops" - maybe you want to print 5 boxes with lids, then 3 small cubes, then a benchy, then four sets of tables made of a top surface and four legs each. In the new setup, you can queue them all up without having to first loop through your boxes (manually keeping track of how many you've printed), then doing the individual prints, then starting another loop with the tables. Queue items and jobs are both draggable, so reordering the queue should also be way easier vs the current plugin method of nudging individual items with the up and down buttons.

LMK what you think, or if you have trouble setting it up. Or if you want to wait for it to get merged into this repo, that's fine too :)

peterson-caleb, 2022-02-14:
Missing this feature is a bummer for me as well. I love the new version of the plugin though overall. 

In order to keep my textured PEI sheets from wearing out in a specific area, I print in 5 different areas. I used to add all 5 to the queue and let them loop through until the filament ran out. Now I need to add them in order to the queue with the lack of the loop function. @smartin015 's idea of jobs would fill my needs really well.

smartin015, 2022-03-13:
Plugin ownership has been transferred, and the job feature is now available in the latest  [1.4.1 release](https://github.com/smartin015/continuousprint/releases/tag/1.4.1 ). If you have any suggestions/find any problems please open an issue at https://github.com/smartin015/continuousprint/issues. Thanks!

closed 2022-03-17 by Zinc-OS