Zinc-OS/continuousprint #58: Updated API?

aburtonProto, 2022-01-08:
The new version ( >= 1.3.0 ) has changed the API.  My plugin called the old resumeQueue function from the API.  This now returns a 404 error.
What is the new API to use for this plugin, or where can I find it?

Thanks.

smartin015, 2022-01-17:
Hi @aburtonProto - `.../resumequeue` is now `.../set_active`, and takes a form entry "active" of true/false. 

You can see the source code at https://github.com/Zinc-OS/continuousprint/blob/master/continuousprint/__init__.py#L218

aburtonProto, 2022-01-21:
Thank you for the /set_active pointer.
Did the [pause] command in the clear bed section get removed?
It is not working on my prints, and I cannot find any reference to it in the code any more.

jarfil, 2022-01-22:
The [PAUSE] command seems to have been lost in commit bc0c7ac9bd8a3ee4d7c8988e01445c1445e341f6.
Seems like the last working version is 1.2.3a (05117c919664995318d81a16f3a71cbba23ef99f).

Not sure what the plan is with that, but I would really like to have the "pause instead of bed clean" feature back.

smartin015, 2022-01-24:
Hey @aburtonProto, @jarfil  - just wanted to give you a heads up since I've been working on this.

Octoprint already provides its own built-in [at-commands](https://docs.octoprint.org/en/master/features/atcommands.html) which includes an `@pause`, making a plugin-custom `[PAUSE]` implementation redundant. It *should* be as simple as adding `@pause` in your clearing script when you want octoprint to wait for manual input.

But it isn't. Two things went wrong here:

1. I didn't test that this worked in in the bed clearing/finishing script part of the plugin. Turns out that Octoprint picks up the command, but there's some logic in Octoprint's default "pause" handler that bails out if the printer's not actively printing a specific file (as happens when we're in between prints). This requires changing how bed clearing is handled (specifically: it needs to think it's running a print file, not just sending a string of gcode).
2. I didn't update the `[PAUSE]` description text in the settings page so folks knew there was a default option.

I have a draft PR which should make at-commands work in the bed clearing/finish scripts, and also better show what's happening as the bed is clearing and when the pause happens. It still needs some cleanup, though. I hope to have it pushed to [my fork](https://github.com/smartin015/continuousprint) later this week, but it's up to @Zinc-OS on whether/when it's merged into the plugin proper.

smartin015, 2022-01-24:
Friendly update: I've got this working in the [prints_are_files branch of my fork](https://github.com/smartin015/continuousprint/tree/prints_are_files) - but I'm waiting on @Zinc-OS to review #62 before I submit another one.

aburtonProto, 2022-01-27:
Been testing these out.
From my original question, the set_active works as I hoped.  
New code for anyone who needs it is:
```
self.setActive = function() {
	$.ajax({
		url: "plugin/continuousprint/set_active",
		type: "POST",
		dataType: "json",
		headers: {
			"X-Api-Key":UI_API_KEY,
		},
		data: {active: true}
	});
}
```
Notice it is now a POST instead of GET, and include active:true in the data.

@smartin015 [prints_are_files branch](https://github.com/smartin015/continuousprint/tree/prints_are_files) works great for using the @pause command in my testing so far.  Thank you, that is exactly what I needed.

piebaker, 2022-01-31:
Just came across this awesome plugin, but just wondering @smartin015 is your public fork [here](https://github.com/smartin015/continuousprint) more stable/feature complete/in more active dev than this one? You don't have issues on so I can't ask there directly.

smartin015, 2022-01-31:
Hey @piebaker! I turned on issues for my fork :)

AFAIK it's more in active development. I'm working on several improvements - including clustering queue items into repeatable jobs and supporting drag & drop reordering - however, @Zinc-OS is the current maintainer of the plugin and intends to continue supporting it (see our conversation in #48).

We're unaffiliated - I've been making changes to improve my own setup and contributing them back here, but Zinc sets the pace as I have no ability to merge my own PRs.

If you want faster features but still want some stability, I recommend installing from master of my fork, at least until we've worked through the backlog.

piebaker, 2022-01-31:
Awesome. Thanks, @smartin015, I'll grab your master fork for use now. Looking forward to those features!

smartin015, 2022-02-07:
To keep things uncluttered, I'm starting a GH issue which I'll update whenever I do a master merge in my fork: https://github.com/smartin015/continuousprint/issues/3

@piebaker you can subscribe to that issue to know when to pull changes. I hope you enjoy the latest ones :)

smartin015, 2022-03-13:
Plugin ownership has been transferred, so you can now get releases in the usual way (vs installing from master). I think there's still some useful discussion to be had re: documenting and providing a clear API, so I opened https://github.com/smartin015/continuousprint/issues/10 and we can follow up there.

closed 2022-03-17 by Zinc-OS