Zinc-OS/continuousprint #65: No error message

TylonHH, 2022-02-04:
Hey,
I have to say that I'm still on 1.2.3a0 because of the missing loop function. What I figured out: `'InvalidFileLocation' is not defined` because of `printfile.gcode does not exist in local storage, cannot select for printing`

How to reproduce:
add file to queue, delete file, start print
nothing will happen, even no errors

If it's fixes in the newest version, just close this issue.

Cheers 🥳

TylonHH, 2022-02-07:
reaction here 👍 for option 1

TylonHH, 2022-02-07:
reaction here 👍 for option 2

TylonHH, 2022-02-07:
reaction here 👍 for option 3

TylonHH, 2022-02-07:
reaction here 👍 for option 4

smartin015, 2022-03-03:
Friendly update: I'm waiting for [this discussion](https://github.com/OctoPrint/plugins.octoprint.org/issues/1021) to shake out before doing much more involved work. 

To move the conversation along a bit, I think I'll go for option 2 as it can also handle users deleting print files directly from the device, whereas option 1 would still break in that case.

smartin015, 2022-03-13:
Plugin ownership has been transferred - I opened https://github.com/smartin015/continuousprint/issues/7 to track further development.

closed 2022-03-17 by Zinc-OS