Zinc-OS/continuousprint #66: queue finished script not kicking off

pimpmytank, 2022-02-09:
I added the following code to the "queue finished script"and it is not kicking off. The functionality I need is to raise the print head after the Q finishes so that the hot end does not stick to the last part printed.

M117 All Print Complete
G91                            ;Relative positioning
G1 E-2 F2700                   ;Retract a bit
G1 E-2 Z20 F2400              ;Retract a bit more and raise Z
G1 X5 Y5 F3000                 ;Wipe out
G90                            ;Return to absolute positioning

M117 Raising Print head - Turning off
G1 X0 Y{machine_depth} ;Ta Da
M106 S0                        ;Turn-off fan
M104 S0                        ;Turn-off hot end
M140 S0                        ;Turn-off bed

M84 X Y E                      ;Disable all steppers but Z

TylonHH, 2022-02-11:
This is my finishing script
`G28 X0; homing axis
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
G92 E1
G1 E-2 F300
G1 X2 Y145 F800
G91
G1 Z10 F300
G90
M84 ; disable motors
M300 S1397 P240
M300 S1760 P240
M300 S2093 P240
M300 S1397 P720
M300 S0 P240
M107 ; Turn off Fan`

pimpmytank, 2022-02-11:
Thanks. it's not that I don't have code to run in that feature, it's just not kicking off when all my prints complete.

TylonHH, 2022-02-14:
I mean in your case you can try separate E and Z moves, like:
```
M117 All Print Complete
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 Z20 F2400 ;raise Z
G1 X5 Y5 F3000 ;Wipe out
G90 ;Return to absolute positioning

M117 Raising Print head - Turning off
G1 X0 Y{machine_depth} ;Ta Da
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hot end
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
```

softwarecrash, 2022-02-18:
same problem here, the clean script works fine, but when queue done, the end script will not started

smartin015, 2022-03-13:
Hey all - plugin ownership has been transferred to https://github.com/smartin015/continuousprint and a new `1.4.1` release published, which includes significant changes to how the scripts are run. I'm not promising it'll work (I didn't actually investigate this problem in `1.3.1`), but I recommend trying again with the latest version as there a chance it may have been fixed implicitly

If you still see problems, please open a new issue at https://github.com/smartin015/continuousprint/issues (i.e. NOT this repository) and link back to this issue for history. Thanks!