Zinc-OS/continuousprint #69: Wait for bed cooldown not consistent between printers.

willindiana, 2022-02-11:
The G-Code command M190 (Wait for Bed Temperature) does not always wait for the bed to meet the target temperature. This is dependent on your printer, and often results in the clear bed routine running before the bed has cooled down fully causing the print to not eject from the bed or at worst damage to the printed part. 

 I have created a PR for a bed cooldown routine that can be set by either a target temperature or a timeout being met. 
This change has been tested by me.

https://github.com/Zinc-OS/continuousprint/pull/46

DanaViolet, 2022-03-10:
@willindiana I installed your fork onto my octoprint yesterday to see if it would fix the issue I was having where my machine would eject the entire build plate after the first print in the queue.... and while it seemed to work, I noticed that the part was ejected about 10 degrees C above the 25 that I was expecting. Could this have something to do with the timeout you set, and my bed not cooling fast enough?

willindiana, 2022-03-10:
Hello @DanaViolet  Could I get a little more context. 

- What is the printer you are using? 
- Did you use my latest branch?
- Are you running any pre-cooldown scripts?
- What was the timeout that you had set? 

If you are using my latest changes you should be able to set the timeout and goal temp independently. When either of those is met it should run the bed clear routine. 

I would try two things.

My first step would be to increase the timeout and try agian.

Thanks for your valuable testing looking forward to hearing back!

DanaViolet, 2022-03-11:
@willindiana 

I'm using a prusa mini with the latest firmware.
I installed the master branch yesterday.
Pre-cooldown scripts? I don't think so.... I left everything on default for testing purposes.
As for timeout, whatever you set for default. I cant remember what it was or where I found it.

Where do I set the timeout? Is there a maximum value? 
I'd prefer to let it wait an hour before ejecting the part, rather than try to eject while the bed is still warm.

willindiana, 2022-03-14:
Hey @DanaViolet  you should be able to find the two fields for target temp and timeout in the plugin settings page. The timeout should be set in minutes. Also FYI for anyone listening any further comments related to this issue should take place here https://github.com/smartin015/continuousprint/issues/6 as that will be the new home of the continuous print plugin and I will be working with that project owner to have these changes added to master.

closed 2022-03-17 by Zinc-OS