Zinc-OS/continuousprint #70: Is this project abondoned?

willindiana, 2022-02-14:
Hey @Zinc-OS , 

I really appreciate your work on this plugin!
I would kindly like to ask if you still wish to maintain this project? 

There are several issues piling up and I have had a PR open since October of last year. https://github.com/Zinc-OS/continuousprint/pull/46

I know myself and @smartin015 would love to be set as collaborators so we can share the workload.

If I don't hear back in 14 days so 2/28/22 I will open an abandoned plugin request with octoprint.

Of course, I would love for you to continue to maintain this plugin if that is what you want!

Respectfully,

Will

smartin015, 2022-02-15:
Agreed - we're happy to collaborate, but it's difficult to contribute bugfixes and improvements when the review cycle is months long.

If you prefer we weren't set as collaborators, I can instead publish my fork under a different name (with attribution of course). Folks can use it for "bleeding-edge" features, but refer back to your plugin for more stability.

TylonHH, 2022-02-16:
https://github.com/Zinc-OS/continuousprint/issues/48

smartin015, 2022-03-13:
Plugin ownership has now been transferred - any future issues should be opened in https://github.com/smartin015/continuousprint/issues.